robomakery/pvc-powerchair #1: change to Arduino processor

dylanvaughn, 2016-05-06:


miliapa, 2016-06-15:
This might help with board selection:
http://makezine.com/comparison/boards/

There are several with built in bluetooth support.

Alternatively, look for a board with a bluetooth shield available.

If your goal is to use ROS, you might want to consider a board capable of running Linux.

bus710, 2016-06-25:
Since Arduino's basic approach doesn't accept timer interrupt, I rather pick other systems. I still need to consider affordable price tags though.

Thus, I am thinking to use STM32F0-disconvery or Mbed since both have pros such as decent price, enough performance, easy to use, and open echo system. Especially, the discovery board will be upgraded to OpenCR, which is a part of real time ROS system soon. (so the middleware in the board will natively exchange ROS message with a x86/Ubuntu based system.)

I want to listen from you about this idea.

dylanvaughn, 2016-06-28:
@bus710 can you give me more details on the Arduino interrupt issue?

Regarding high-level board design, I think we should have a micro linux computer (I have a [snickerdoodle](https://www.crowdsupply.com/krtkl/snickerdoodle) coming soon which we could use for this - or even something like the [Intel NUC](http://www.intel.com/content/www/us/en/nuc/overview.html)) which runs ROS and sends control instruction to arduino boards which interface with the motor drivers, etc. via the [ros_arduino_bridge](http://wiki.ros.org/ros_arduino_bridge).

Maybe we should make a diagram outlining the board setup - something like this:

https://lh3.googleusercontent.com/-_g2DF_fLgn4/V26gBZy1c4I/AAAAAAAAADA/8IWbXK99Ebge-WIxoAY7ONzaaNgUJDezACLcB/s1600/schema_JQR_5.jpg

bus710, 2016-06-28:
1.
as you know, polling may cause timing issues to run motors.

if we use too many motors over than arduino's maximum PWM output, we have
to drive one or more motors by using timer or polling. in this case,
polling can make some delay. (if we use arduino mega, that's nice to have.)
however, i agree that arduino can be the best approach for the first step.

2.
i wish we can use NUC to avoid compatibility issue in drivers (sensor,
network, and so on.) but we also need to think about power source.

3.
yes, let's talk about a rough diagram.

SJ

closed 2017-05-03 by dylanvaughn