robomakery/pvc-powerchair #10: Use electro-mechanical circuit breaker instead of automotive fuse

dylanvaughn, 2016-05-06:


SBerneberg, 2016-05-17:
I have used these auto reset breakers with great success.  I usually mount them on the battery; a convenient location.  Here is a picture:
[image]

SBerneberg, 2016-05-17:
I usually get these in El Cajon for about $2.25 for a 30amp breaker.

dylanvaughn, 2017-05-03:
Find sources for these parts, purchase, and integrate into drawing and prototype v2:

* electro-mechanical circuit breaker instead of automotive fuse
* on / off switch
* charger port
* trickle charger adapter

dylanvaughn, 2017-07-07:
Still left to buy:

* Custom Arduino shield
* Voltage divider
* Fan
* JST connectors
* Power plug
* Power jack
* Hose Clamp

dylanvaughn, 2017-07-09:
@rbeetem for the custom arduino shield should I get this one:

http://www.jameco.com/z/A000082-Arduino-Arduino-Uno-Proto-Shield-PCB-only-_2152438.html

which is just the bare PCB only, or this one:

http://www.jameco.com/z/A000077-Arduino-Arduino-UNO-Proto-Shield-Rev-3-assembled_2152391.html

which has a couple of things already set up?

dylanvaughn, 2017-07-09:
@rbeetem is this what I want?  It is prime :)

https://www.amazon.com/Arduino-Proto-Shield-R3-Assembled/dp/B007QXTRNA

rbeetem, 2017-07-18:
The bare PCB is the one I would recommend. 
http://www.jameco.com/z/A000082-Arduino-Arduino-Uno-Proto-Shield-PCB-only-_2152438.html

The only one is designed to have additional shields placed on top, which we currently have no desire to do.

dylanvaughn, 2017-07-22:
@rbeetem I'm planning on ordering this today - does it look right?

https://www.amazon.com/gp/product/B01J1KM3RM/ref=ox_sc_act_title_2?smid=A2OVMUS055KMW5&psc=1

rbeetem, 2017-07-23:
@dylanvaughn Looks reasonable to me.

closed 2017-08-02 by dylanvaughn