robomakery/pvc-powerchair #2: Bigger motors

dylanvaughn, 2016-05-06:


dylanvaughn, 2016-05-09:
from @SBerneberg 

Potential new motors: 

http://www.allelectronics.com/make-a-store/item/dcm-563/12vdc-right-angle-gear-head-motor-with-worm-drive/1.html

very powerful, right angle (left angle is available), 12VDC, 1.3A MAX.  The robots that the students use these motors in typically draw about 500mA.  Approximately double what the current motor draws, but with a 9AHr battery, that is about 9 hours of CONTINUOUS operation.

SBerneberg, 2016-05-17:
This motor has a worm drive for moving a car seat.  We had to remove and modify the worm drive.  Was not that difficult but you will need a lathe to modify the worm drive.  Again, not that difficult but not everyone has a lathe.

dylanvaughn, 2016-05-18:
@SBerneberg I would prefer not to have to modify the motors - are there some brushless motors we could look at?

AdamTeays, 2017-05-17:
So looking at what we need motor-wise, I ran some calculations based on two criteria: that the chair must be able to go up a 7-degree ramp with a 50 lb. passenger, and that the chair can travel at around 5 mph (typical speed of an electric powered wheelchair).

So, first off, I went ahead and calculated what the stall torque would be, which came out to be 9.14 lb-in (10.53 kg-m, or 146.24 oz-in).

Onto determining the motor speed: since the wheel diameter (6 inches), the motors would need to run at ~280 rpm to reach 5 mph. If we want the chair to move 3 mph up an incline, that would be ~168 rpm.

I assume that some of us are familiar with how DC motor specs are calculated. If not, here is a helpful link about that (where specs are typically the stall torque, no load speed, and maximum efficiency).
[http://www.micromo.com/technical-library/dc-motor-tutorials/motor-calculations](url)

[image]

All we need, is a motor that has the point (9.14 lb-in, 168 rpm) at or below that blue line.

Calculation proofs: [https://docs.google.com/document/d/1mss5ZFxsw8eWR4RWg8JPl5gAOq1itzg9Wl9rDcymRVQ/pub](url) (ignore the little question marks, those were supposed to be the symbol for degrees)

Now to actually find a 12-volt DC motor that fits this criteria.

AkiKono, 2017-05-18:
> tutorials/motor-calculations <http://url>
> 1mss5ZFxsw8eWR4RWg8JPl5gAOq1itzg9Wl9rDcymRVQ/pub <http://url> (ignore the

Hello,

Thank you so much for sharing :) This is a good input :)

I was planning to report what I have done so far to the team at the next
meeting, but since there is an attention to motor torque and specs, I guess
I should also share the updates with the team.

I have been using a google sheet I created to calculate torque on the motor
in various situations. Here is the link.
https://docs.google.com/spreadsheets/d/12mFd3bs58vSQ7STA4lI8He2xesg1GwQOK_4B4z5c2mo/edit?usp=sharing

And attached is a video showing how much current the motor is using while
it is under the load.
I am thinking to test the motor with a known external load. (not just
applying random unknown loads using fingers like in the video. haha)

I have been working alone on this motor and motor control since SJ left,
but I am always open to working with people who are interested in :)

Anyway, more detailed report will be ready in the next meeting.

Thanks :)
Aki

dylanvaughn, 2017-11-08:
need to get this motor working again

dylanvaughn, 2018-01-06:
Hub motor option:

http://www.uumotor.com/store/31-8-inch-small-motor-no-hall-sensors-for-electric-scooter.html

dylanvaughn, 2018-02-22:
this is the motor selected by the Capstone team:

https://www.aliexpress.com/item/24V-36V-48V-8Inch-Electric-Wheel-Hub-Motor-350W-Brushless-Non-Gear-Hub-Motor-For-Electric/32837818637.html?traffic_analysisId=recommend_3035_null_null_null&scm=1007.13338.80878.000000000000000&pvid=09e0fc0a-6677-4663-8c43-2f5e89c9a1ac&tpp=1

closed 2018-06-25 by dylanvaughn