robomakery/pvc-powerchair #16: Improve battery / charging

dylanvaughn, 2016-05-27:
Current battery is difficult to access / charge.  Come up with alternate battery that is easier to charge (but does not use inductive charging).

dylanvaughn, 2016-06-15:
Look into using electric vehicle battery chargers / components

johanisforever, 2016-08-24:
I have attach a propose "inexpensive and easy battery charging design". The LEDs at the end are optional, however the "ON" LED will draw power from the battery, the power draw is negligent but during long period of inactivity (weeks) with the switch on the "ON" position the power draw would have to be taken into consideration. 
  
[changing Ckt_1.pdf](https://github.com/robomakery/pvc-powerchair/files/433810/changing.Ckt_1.pdf)

I did not mention the require conductor size, but the minimum conductor size require for power carrying connection needs to be 14 gauge.  Assuming a 200Watt motor     (200Watt / 12 Volt = 16.66Amp)  See bellow 
[image]

closed 2017-11-15 by dylanvaughn