robomakery/pvc-powerchair #20: Gazebo simulation

dylanvaughn, 2016-06-15:
basic gazebo simulation

dylanvaughn, 2016-06-15:
@bus710 to provide more details

bus710, 2016-06-16:
I am thinking to use Onshape to draw a new design for simulation so that I was looking for export option for DAE file. Althohg Onshape doesn't support DAE, STL is in the list. Also fortunately, There are bunch of apps and services, which provide converting between STL and DAE. 

I guess I might try drawing some design by using Onshape and make some shapes in Gazebo this weekend.

dylanvaughn, 2016-06-17:
@bus710 it probably makes sense to just go ahead and make a URDF model for the chair to use in Gazebo - we are already tracking that task in #13.  I put that into this milestone.

bus710, 2016-06-17:
@dylanvaughn, that's true... I am editing one of my previous model to adjust for Gazebo 7. (Because the sdf was design for Gazebo 1.7 and makes some crash for Gazebo 7.)

bus710, 2016-06-18:
Although I just put three objects (body and two big wheel) yet, I think that I can build the first model in this weekend.  

[image]

LeviLister, 2016-06-18:
Looks like a good start!

bus710, 2016-06-20:
I issued a PR for simulation to this repo.
Please let me know if I can merge my branch.

dylanvaughn, 2016-06-20:
@bus710 I left some comments on the PR

bus710, 2016-06-21:
@dylanvaughn, Yes, I saw your mention and it was very sincere. I will fix the PR soon.. Thank you!