robomakery/pvc-powerchair #24: diagram

dylanvaughn, 2016-09-28:


dylanvaughn, 2017-03-08:
[V1_5 Drawing (1).pdf](https://github.com/robomakery/pvc-powerchair/files/826528/V1_5.Drawing.1.pdf)

Here is the existing drawing that @SBerneberg did

dylanvaughn, 2017-04-05:
need to merge all drawings into a draw.io document.
include wires, screws

dylanvaughn, 2017-05-07:
WIP diagram can be found here:

https://github.com/robomakery/pvc-powerchair/blob/develop/docs/pvc-chair-component-diagram.png

dylanvaughn, 2017-06-07:
Need to:

* Integrate terminal block
* On/Off switch
* Arduino Protoshield
* add wire sizing to diagram

dylanvaughn, 2017-06-07:
Also:

* Charging Connector
* Charger
* Switch

dylanvaughn, 2017-06-11:
Lots of progress made:

https://github.com/robomakery/pvc-powerchair/blob/develop/docs/pvc-chair-component-diagram.png

Still have some TODOs but I think it is good enough to start building against.

rbeetem, 2017-06-20:
Updated with Arduino interfaces and pin list. 

pvc-powerchair/docs/pvc-chair-component-diagram2.png

Analog pins are in short supply if we use I2C for the OLED, could use SPI on the digital pins instead.

dylanvaughn, 2017-06-20:
Looks good!  Can you also update the xml diagram source in github?

rbeetem, 2017-06-25:
Updated png and xml in github.

dylanvaughn, 2017-07-07:
last item to do on this is to rearrange diagram and add final wires to/from terminal block

closed 2017-09-06 by dylanvaughn