robomakery/pvc-powerchair #30: prototype electronics + motors

dylanvaughn, 2017-03-08:
solder and document build

JamesNewton, 2017-03-09:
I noticed that the [Arduino UNO](https://www.arduino.cc/en/main/arduinoBoardUno#techspecs) has both 5 volt and 3 volt regulators on board, and the 5 volt one is rated for 	7-12V so as long as we just use 12 volt batteries, we should be ok with that. My [BOB breakout board](http://techref.massmind.org/techref/microchip/bob.htm) has terminals which make it easy to hook up to that (no need for the barrel connector used on the Arduino). If we used higher voltages, the regulator option on my BOB can manage up to 48 volts and then the Arduino could produce 3V. 

I'm only mentioning the 3 volt supply because it's needed to run an [ESP-8266](http://techref.massmind.org/techref/esp-8266.htm) if we decide to add the [web interface](http://techref.massmind.org/techref/ESP8266/WebSerial.htm).

The BOB also supports a small LCD display, and I think we should be able to get it to work on the UNO while still providing signals for the drivers, but it may be that the LCD won't update when moving. No great loss. It's mostly for monitoring voltage levels and providing a UI to change motor response. I plan to wire that up in the next few days just for testing. I did something like it already for my [Reagent Robot](https://hackaday.io/project/4197-reagent-robot), so I'm sure it will work, I'm less sure it will be useful... let me know what you think.

Based on feedback at the meeting on the cost of H-bridge drivers, I've decided to lower the price on [my h-bridge kit](http://techref.massmind.org/techref/io/motor/DRV8701s.htm). It's now under $10 (and, of course, I'm donating 2 to the project). I STRONGLY doubt that the cheap drivers available from China on ebay can actually manage the power they are advertizing, and I KNOW that mine can, but it is a kit, you must solder it together, and so making it cost competitive is... necessary. The main reason I want to use it is that it directly connects to the BOB with a simple cable, there are no wiring issues, and again, I KNOW it works. I plan to solder up a couple before the next meeting so we won't lose any time with that. If someone else wants to build a set with their own driver, that's great!

That's all for now.

tnahcrem, 2017-03-09:
I think we need to add Steve, Oscar to this thread to do the scheduling for our next meet. (unless we plan to do the scheduling via another channel).

As for my availability; I am out of country from 3/11 - 3/19 for business travel. 
I can be available after during evenings our a couple hours on weekend.

dylanvaughn, 2017-03-09:
@tnahcrem I think the best place for general discussion is the Google Group:

https://groups.google.com/forum/#!forum/robomakery-pvc-chair

@SBerneberg is already on the list, but @JamesNewton and Oscar (and yourself) would need to sign up.

tnahcrem, 2017-03-09:
Thanks. I have joined the group.
@JamesNewton  Can you also join the group. I will try to get in touch with Oscar via text and get him in too.

dylanvaughn, 2017-05-03:
We got the basic electronics and motors working in the latest build meeting - closing this

closed 2017-05-03 by dylanvaughn