robomakery/pvc-powerchair #31: research arduino code for v2 control

dylanvaughn, 2017-03-08:
Are there existing libraries for controlling motors with a joystick?

Document functional requirements for software and technical approach.

In future state arduino may communicate joystick commands to odroid to do high level control.

dylanvaughn, 2017-03-08:
Oscar + Husein

tnahcrem, 2017-03-09:
Hi Dylan,
Can you please grant me access to the project ?
Thanks,
husein

dylanvaughn, 2017-03-09:
@tnahcrem done!

tnahcrem, 2017-03-09:
This is a good/easy tutorial to get started with the joystick on arduino.
https://www.arduino.cc/en/Tutorial/JoyStick

Also attached zip contains  refrence document from a Joystick vendo(parallax)
[27800-2-Axis-Joystick-KickStart.zip](https://github.com/robomakery/pvc-powerchair/files/832232/27800-2-Axis-Joystick-KickStart.zip)

miliapa, 2017-03-10:
https://www.engineersgarage.com/contribution/joystick-controlled-dc-motor

tnahcrem, 2017-03-10:
@miliapa  Thanks for sharing. It looks like I have to register to get access to source code ? If you have access can you share the code with us ?

johanisforever, 2017-03-17:
A few days I spoke to Husein, and we agree that we need to start developing the program base on
the following code. We will use what we can.

https://www.arduino.cc/en/Tutorial/JoyStick

To code and test I built a prototype. I will be passing it to Husein once he comes back from his trip since I also will be going on a trip. We will be using Arduino. I will attach pics of the development platform, don't laugh it is just a development board.
Thing to keep in mind:
-I included a diagram of the current set up.
-I included a list of parts of the current set up and the required parts for the final set up.
- I include some pushbutton and light so we can code it in a way that when completed the PWM can be adjusted easy for different motors.  Different motor may need different PWM frequencies for optimum performance. Also, high RPM with low gear motors will need shorter duty cycles to prevent the motor from spinning too fast.
-currently I chose a  10 Amp h-bridge to power small motors, the final product will require a higher Amp h-bridge and higher wire gage.
- The 12Volt battery for the current set up may be too heavy, so I may have to swap out batteries.
-IMPORTAN!!!!! use the LEDs to first test the both the  PWM and the break signals before making the final connection the H-bridge. Giving both a commands, forward and back, at the same time will result in a short that will destroy the h-bridge. 
-To prevent any damage a fuse of 8 amp will protect the entire circuit. 
- as far as PWM I would like to start with 25% duty cycle, then see how fast the motor rotate, if they are too slow we can increase the duty cycle until we determine the max allowable speed. once we agree on the max allowable speed for a particular motor, we can divide the duty cycle by 5 increments. I say by 5 because I believe that 5 speeds would be enough for the wheel chair,
Example:If a reasonable fast speed is attained at 50% duty cycle. then 40%, 30%, 20%, 10% duty cycles will control the slower speeds, naturally 0% would be stopped. 
Now this idea may, or may not work, but once the code be completed, we can experimentally determine if those are good values.  
a different motor will require different max speeds, but we can write the code with options to set the max speed via push buttons. For safety we could put a requirement that to enable the push buttons the user would have to temporary jumper a particular input to a particular output and so entering the "setting mode" once the jumper is removed, accidental pressing of any button would not have any effect. 

I did not included the charging circuit yet, but any trickle charger will do.

Note to all, my schematic does not supersede the V1_5 Drawing.pdf submitted a week ago, rather it complements it. 
[image]
[parts.docx](https://github.com/robomakery/pvc-powerchair/files/851821/parts.docx)

johanisforever, 2017-03-17:
this are the images of the "development platform"
[image]
[image]
[image]
[image]

closed 2017-05-05 by dylanvaughn