robomakery/pvc-powerchair #32: Contact Dan Hendricks at OSML

dylanvaughn, 2017-03-08:
Investigate getting a group membership that we can use for builds.

JamesNewton, 2017-03-09:
dan@opensourcemakerlabs.com

dylanvaughn, 2017-03-09:
@SBerneberg @JamesNewton what tools do we need for the build?

JamesNewton, 2017-03-09:
For the electronics only:
Soldering Iron, solder, workbench, good light. 
Basic hand tools (pliers, screwdrivers, bias cutters, strippers, etc...)
Multimeter, scope would be nice for troubleshooting.
PC with USB port and Arduino IDE installed or internet connection to install it.
and one other tool which it absolutely critical and won't be thought of until we can not proceed without it.

I have all that stuff here in my office in Escondido, but it's rather cramped, and could really accommodate only one or two other people. 

I should just build it and video? I'd really rather work with someone... Maybe Steve or someone else could stop by.

dylanvaughn, 2017-03-09:
@JamesNewton I would prefer to build it together as well.  My garage might be a good option.  I have basic electronics and other tools and room for probably six people comfortably.  It is in PB however :).

I also have emailed OSML to see what options we may have there.

JamesNewton, 2017-03-09:
I checked and PB is basically an hour away from Escondido on a weeknight, but it's about 40 minutes on a weekend. I'm open to driving to PB on e.g. 3/18 afternoon and I could bring others from NC or we could meet up. e.g. Steve could drive over from Spring Valley and meet me at my place, then we could go on and pick up people in SM or meet people from Vista and points west at a park and ride before heading south down the I-15. Perhaps someone costal could do the same from Vista.

At the same time, I'm liking the idea of just building a basic version here and now, to test and debug where I have all my tools, maybe with Steve or a very small group of those who are really interested, and then meeting up in PB or wherever to install it in the chair or onto the real motors, etc... that I don't have here. We could do that this weekend, e.g. 3/11 or 3/11 in the afternoon, or one evening next week (except Wednesday). We could record and present what we did, and possibly re-build another set at the main meeting if that has value.

dylanvaughn, 2017-03-30:
We had a build meeting at Dylan's garage and it worked fine as a location - I'll plan on hosting at least one build meeting per month (in addition to the planning meeting).

closed 2017-03-30 by dylanvaughn