robomakery/pvc-powerchair #34: BOM in editible format?

JamesNewton, 2017-03-16:
Would it be possible to transition the new BOM to something editable like a text document (.CSV, .TSV) or a google sheet so that those of us without Excel can edit it?

https://github.com/robomakery/pvc-powerchair/blob/develop/docs/PVC%20Parts%20List-V2.xls

Github plays nicely with .CSV:
https://help.github.com/articles/rendering-csv-and-tsv-data/

If that is acceptable (e.g. no one objects) I'll just do it.

dylanvaughn, 2017-03-16:
@JamesNewton the CSV support for GitHub is neat - I haven't seen that before.  It would be nice to have the BOM in the source code (as a csv).  I would vote for that or Google Sheets.

JamesNewton, 2017-03-16:
Ok, it's there as the csv. I updated it with links to the parts I'm donating, and tweaked the formatting a bit to fit better on the page. Honestly, it's a bit of a pain to edit as a CSV file, but it is nice to have the BOM in source.

closed 2017-03-16 by JamesNewton

dylanvaughn, 2017-03-16:
I like it!