robomakery/pvc-powerchair #35: CADify Joystick Enclosure STL and add bottom cover with attachments

dylanvaughn, 2017-03-30:
Current part:

https://github.com/robomakery/pvc-powerchair/blob/develop/hardware/printed_parts/JoystickEnclosure.STL

Needs to be created in OnShape.  We also would like to add a bottom plate which can be screwed into this piece.

dylanvaughn, 2017-05-03:
Update from planning meeting:

* research if we should redesign this piece

dylanvaughn, 2017-06-07:
Since we don't have the old joystick board we can make this lower profile and add a lower piece.  Still planning on 3d printing this piece.

dylanvaughn, 2017-09-06:
for v2 we are going to print the part as is

badiozam, 2017-09-06:
Hey Dylan, do you have the direct link to OnShape? I found https://cad.onshape.com/documents/afa971ef5507b38d69394b06/w/1b6f32bd7bb64483d9a5f636/e/bac4a40611c0be1aa8d12fcc but I don't think that's the right one because it doesn't have the part on there.

McCAndrew88, 2017-09-07:
I can share it with you, what is your email?

badiozam, 2017-09-07:
Hey Andrew, It's my github username at yahoo. Thanks!

-Hamid

McCAndrew88, 2017-09-07:
You should have access to it now.

The JoystickEnclosureV2 is the one I modeled after what we have and you
should be able to export an STL out of that.

V3 is the lower profile one, but the bottom tabs don't work well. The
bottom still needs design tweaks.

https://cad.onshape.com/documents/4aca10982c02d999f4bbff8b/w/3f4c62dba35cde9e394a6221/e/2ed27eae5eff1055e6bc6cef

You can edit anything, just makes sure to make a new version.

Let me know if you have any questions!

badiozam, 2017-09-07:
Awesome, thanks!

-Hamid

badiozam, 2017-09-08:
I went ahead and pinted the part out and I was able to get it to fit onto a 1/2" PVC I had laying around.
[image]

However, for some reason the outer ring came out a little warped:
[image]
[image]

I think I can get rid of it if I adjusted the support threshold in my slicer settings but I wanted to ask if you think it's acceptable as is.

dylanvaughn, 2017-09-08:
Looks good to me!

closed 2017-10-11 by dylanvaughn