robomakery/pvc-powerchair #36: CADify Motor/Gearbox plate and machine

dylanvaughn, 2017-03-30:
We need to create an OnShape CAD / STL version of this:

http://www.openwheelchair.org/Products/GearBox-Plate/

Then we should talk about whether we want to stick with aluminum or try acrylic (which we could cut at MakerPlace)

McCAndrew88, 2017-03-31:
I can get on this as soon as I get my hands on the part.

dylanvaughn, 2017-04-05:
plan to use acrylic for v2 and laser cut at makerplace

McCAndrew88, 2017-04-10:
Finished the motor plate and 3D printed a thinner version of it to make sure it would be good tolerances. Seems good, and CAD is up to date. Should I look into getting it made this week?

dylanvaughn, 2017-04-10:
Do you have access to a laser cutter? If so that would be great. I was
planning on renewing my membership to makerplace to get it cut...

McCAndrew88, 2017-04-10:
> 292845807>,

No I do not have access to a laser cutter, so your makerplace will probably
be the best option.

What do you need me to provide you to get it cut on the laser cutter?

dylanvaughn, 2017-04-10:
> 293006020>,

OK sounds good - I forget the process for generating the appropriate file
but I think I just need an STL to start, which I will get off of OnShape.

dylanvaughn, 2017-04-10:
Motor plate in OnShape is here:

https://cad.onshape.com/documents/e45c7a3e9ffce924c9f8be0e/v/b85c2a483f26aa34e6aeb54a/e/d49b3f0c8c964535a9f3ce61

McCAndrew88, 2017-04-10:
> Motor plate in OnShape is here:
>
> https://cad.onshape.com/documents/e45c7a3e9ffce924c9f8be0e/v/
> b85c2a483f26aa34e6aeb54a/e/d49b3f0c8c964535a9f3ce61

Yep, that is it. Right now V1 is set to the 3.3mm thickness of the aluminum
so to make it more accurate, let me know what thickness of acrylic we want
to use.

dylanvaughn, 2017-05-03:
Make new issue to track sourcing this piece (see #40)

closed 2017-05-03 by dylanvaughn