robomakery/pvc-powerchair #37: Research and purchase motor driver

dylanvaughn, 2017-04-05:
example:

http://www.ebay.com/itm/60A-High-Power-MOS-Dual-Channel-H-bridge-DC-Motor-Driver-Module-/181947954406?hash=item2a5cf174e6:g:UDgAAOSwf-VWXXxI

Find similar products on roboshop and / or sparkfun

Send selection to mailing list, then purchase

dylanvaughn, 2017-04-07:
Another option:

https://smile.amazon.com/Sabertooth-Dual-12A-Motor-Driver/dp/B00CT5CHIC/ref=sr_1_3?s=hi&ie=UTF8&qid=1491587619&sr=1-3&keywords=dual+channel+motor

dylanvaughn, 2017-04-07:
Or this:

https://www.amazon.com/Sabertooth-Dual-25A-Motor-Driver/dp/B008OMQUXC

SBerneberg, 2017-04-07:
I am re-thinking this whole motor driver thing.  The motor drivers with more drive capability are very expensive.  We could purchase a couple of the drivers with lesser current drive than one with more current drive.  I would say get the one that we currently have, and get it working and take a chance that we will not have rotor lock that will last very long.  Let's spend the resources on a better solution like the brushless approach.  I would however put some heatsinks on the transistors so we have a better chance of surviving a rotor lock.

SBerneberg, 2017-04-07:
Sorry, didn't mean to close this...still learning.

dylanvaughn, 2017-04-09:
@SBerneberg I bought this one:

https://smile.amazon.com/Sabertooth-Dual-12A-Motor-Driver/dp/B00CT5CHIC

before I got your recent messages :smile: 

Let's use the one I bought for v2 - things are going to change for v3 b/c of the BLDC motors anyway.

closed 2017-04-09 by dylanvaughn