robomakery/pvc-powerchair #40: Cut motor plates

dylanvaughn, 2017-05-03:
Research either getting the plates cut at makerplace or using an online service (a la Shapeways) that will cut the parts for us.

McCAndrew88, 2017-05-05:
Getting these cut through customlasercutting.com which is actually Pololu Laser Cutting. There is a minimum order on these so I am getting 3 motor mounts. They come next Wednesday and hopefully match well with the gear boxes. I'll close this issue after I know they will work.

McCAndrew88, 2017-05-12:
Motor mounts are cut and fit nice!
[image]

dylanvaughn, 2017-05-12:
@McCAndrew88 Looks good!  What was the cost?

McCAndrew88, 2017-05-12:
$33.45 for 3 after shipping and everything. They did a really good job and it was quick, they just had a minimum so I had to order 3. Still shopping around for other places, will see what I come up with once I finish the bottom plate and box.

dylanvaughn, 2017-06-07:
completed

closed 2017-06-07 by dylanvaughn