robomakery/pvc-powerchair #41: Battery level indicator

dylanvaughn, 2017-05-03:
Research battery level indicator attached to arduino.

When battery is low, turn on LED.

dylanvaughn, 2017-05-03:
Assign to Rob once Dylan gets his github username.

rbeetem, 2017-05-07:
github user name rbeetem

rbeetem, 2017-05-23:
Basically a voltage divider. Two known value resistors fed to Arduino analog input. Arduino input is 0-5v, 10 bit resolution. 

Implementation Example: 
http://henrysbench.capnfatz.com/henrys-bench/arduino-voltage-measurements/arduino-25v-voltage-sensor-module-user-manual/

Add logic for LEDs in Arduino code. 

TODO: number of LEDs desired? (1 red) or (1 red and 1 green) or (1 red, 1 yellow, 1 green) or anything else.

dylanvaughn, 2017-06-07:
Need to identify parts and add to the BOM and drawing

dylanvaughn, 2017-06-07:
Let's go with a OLED display:

https://www.google.com/search?q=oled+64x32&source=lnms&tbm=isch&sa=X&ved=0ahUKEwiRorjnyarUAhUrxVQKHWkBB4gQ_AUIBigB&biw=1600&bih=1097#imgrc=_

closed 2017-07-07 by dylanvaughn