robomakery/pvc-powerchair #44: Clamp for reclining pipe

dylanvaughn, 2017-05-03:
Currently the pipe that controls the reclining of the seat is loose.  Research how to tighten this once the desired decline is reached.

Ideas:

* Hose clamps
* Rubber o-ring

dylanvaughn, 2017-07-07:
@SBerneberg tried the two above options and determined they will not work - need other ideas

maybe:

* Drill holes in pipe and use rod to stop movement
* ???

@McCAndrew88 ideas?

dylanvaughn, 2017-07-07:
Maybe this:

https://www.amazon.com/gp/product/B01MF4M54W/ref=s9_acsd_zwish_hd_bw_bHLZE3_c_x_w?pf_rd_m=ATVPDKIKX0DER&pf_rd_s=merchandised-search-12&pf_rd_r=C917DN5QWRM99K24YGED&pf_rd_t=101&pf_rd_p=57fcc92e-05c1-59d1-acc7-7eb62e121007&pf_rd_i=256338011

McCAndrew88, 2017-07-07:
Those hose clamps seem like they would work. What was the problem with them?

If those don't work, I like the idea of using a rod to stop the movement like machines at the gym. I can prototype something for that idea.

dylanvaughn, 2017-07-09:
@McCAndrew88 no issue with the hose clamps - I'm going to buy those to try.

closed 2017-10-11 by dylanvaughn