robomakery/pvc-powerchair #48: Cut bottom plate

dylanvaughn, 2017-07-07:
Cut the bottom plate (which will attach to the casters) as shown in the OnShape model

McCAndrew88, 2017-07-07:
Before we do this, can we get the updated width of the current prototype?

Mainly, the length of the PVC pipes running across the seat and back. I can create a new version of the assembly that is to the current specs so we can cut the correct lengths of the back.

dylanvaughn, 2017-07-24:
@McCAndrew88 the numbers are in the 'PVC_2' tab of this spreadsheet:

https://github.com/robomakery/pvc-powerchair/blob/develop/docs/PVC%20Parts%20List-V2.xls

Please let me know if you want me to verify some of the measurements by hand.

dylanvaughn, 2017-08-02:
@McCAndrew88 we took some measurements while looking at the CAD model and think the piece of wood should be 14.25" x 7", with wheel cutouts down to 11.5" x 4.75".

McCAndrew88, 2017-08-02:
@dylanvaughn I am going to update the cad model with the new specs that you sent. The current is the original model and much smaller. I will get the dimensions from that and look for a place to cut the bottom unless someone has a way to cut them?

Also, the way it is set up, if we want to move the box to the middle, we will have to make some kind of spacers for the casters.

dylanvaughn, 2017-08-29:
@McCAndrew88 I plan to cut the wood at my house this Thursday with @SBerneberg - could you let me know if I am good to get the dimensions from the CAD model or should I look at the current prototype under construction?

McCAndrew88, 2017-08-29:
The model is up to date with all the dimensions in the spreadsheet you sent so it should be good!

closed 2017-09-06 by dylanvaughn