robomakery/pvc-powerchair #63: Research encoders needed with hub motor and oDrive

dylanvaughn, 2018-01-10:
https://discourse.odriverobotics.com/t/which-encoders-to-choose/63

SBerneberg, 2018-01-10:
The comments in this section are correct.  I understand the off-axis encoding but you have to make sure that you compensate for the off-axis.

 

Spatial Applications
sberneberg@spatialapplications.com
(760) 749-9660

This message is intended only for the use of the individual or entity to which it is addressed and may contain information that is privileged confidential and exempt from disclosure. If the reader of this message is not the intended recipient or an employee or agent responsible for the delivering the message to the intended recipient, you are hereby notified that any dissemination, distribution or copying of this communication is strictly prohibited. If you have received this communication in error, please notify us immediately by telephone and return the original message to us by mail. Thank You.

**********************************************************************

This message and any attachments may contain legally privileged and/or
sensitive information. Any unapproved disclosure, use or dissemination
of this e-mail message or its contents, either in whole or in part, is
not permitted. If you are not the intended recipient of this e-mail
message, kindly notify the sender and then securely dispose of it.

**********************************************************************

On Wed, 10 Jan 2018 02:59:38 +0000 (UTC), Dylan Vaughn  wrote:
 
https://discourse.odriverobotics.com/t/which-encoders-to-choose/63

dylanvaughn, 2018-02-07:
potential product with good motors:

https://www.amazon.com/Segway-miniLITE-Balancing-Transporter-Integrated/dp/B074CYX5PT/ref=sr_1_3?ie=UTF8&qid=1517970785&sr=8-3&keywords=segway+mini+lite

dylanvaughn, 2018-06-24:
We are going to use these encoders for v4:

https://odriverobotics.com/shop/cui-amt-102

closed 2018-06-24 by dylanvaughn