robomakery/pvc-powerchair #66: Buy makerpipe connectors and EMC

dylanvaughn, 2018-02-22:
Also need the heat shrink for color

dylanvaughn, 2018-02-22:
@AkiKono could you please put in the makerpipe connector information we need to build the original frame?

AkiKono, 2018-02-22:
Here is the number of connectors we need to build one wheelchair. (v2)

PVC WheelChair        MakerPipe 
Connector A         =     90 degree connector       = x10
Connector B         =     ------
Connector C         =     90 degree                       = x6
Connector D         =     90 degree connector      = x6
Connector E         =     T-Connector                   = x2
Connector F         =     T-Connector                   = x7

In total, 90 degree connector  x22 and T-connector x9.

The maker pipe connectors' length is different from that of PVC pipe connectors, so the length of the  electrical conduit pipes might be different from the length of the PVC pipes.

dylanvaughn, 2019-06-05:
this was done

closed 2019-06-05 by dylanvaughn